78/xiaozhi-esp32 #1052: Sometimes no audio response, text is shown

snooopy3652, 2025-08-08:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.8.5 different devices

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

Since updating to 1.8.5 the devices (1.28 box, 1,54 mumba and 面包板（WiFi）) sometimes do not put out audio. 
This is in two different situations: directly after start or after long time of no conversation.
At first I thought the device doesn't here me, but checking the display it also shows the answer.
The screen will not change and the device does not speak.
pressing the response button does often work and it will start speaking.

I had this issue directly after first boot with 1.8.5, on all 3 devices, so I'm confused nobody here had a similare issue?

### What is the actual behavior?

No audio output, text stuck on screen

### Steps to reproduce.

1. Start device or wakeup device after long "sleep"
...

### Debug Logs.

```plain
is it saved somewhere? Or do I have to use usb data connection while the device is running?
```

78, 2025-08-08:
1.8.5 is frequently reported unstable and 1.8.6 is on the way.

snooopy3652, 2025-08-08:
Thank you, didn't consider this as "unstable", but I will wait for 1.8.6,

closed 2025-08-08 by snooopy3652

78, 2025-08-09:
I just uploaded the new version: https://github.com/78/xiaozhi-esp32/releases/tag/v1.8.6