78/xiaozhi-esp32 #1056: Unable to retrieve activated device on account

Siderate, 2025-08-08:
Hello, 

I purchased ESP32-S3-Touch-LCD-3.5-C product, I successfully flashed it and registered it on the Xiaozhi.me platform.

However, when I entered again on xiaozhi.me, the "device agent" is nowhere to be found, and I can't find the device at all on my account, which means that I can't configure it anymore.

It asks all the time to add a new device (with new code), like I'm unable to link the account to the device. I can still talk to the AI, and it remembers the old parameters, but can't change any parameter since it doesn't appear to be linked.

[image]

All the time it asks for a new code, but the device has already been registered today. Can you please help me, thank you very much.

78, 2025-08-09:
I believe your device is already linked to another account. If you want to recover your account or re-link your device, please email **xiaozhi.ai@tenclass.com** for assistance.

Siderate, 2025-08-09:
Thank you very much for your answer, I bought this device only to be able to use xiaozhi. 

It's much appreciated, I sent already a mail, hope I'll receive an answer.

Thank you again 🙏

closed 2025-08-12 by Siderate

Siderate, 2025-10-31:
Yes, don't worry, just drop them a mail and mention your device and mac address of the device. 
They will unlink the device so you'll be able to link it again.

makeriocc, 2026-02-26:
> Hello, It's been a long time, I forgot which firmware I used but I remember trying both: either way, if you used a wrong one you should be able to reflash your device again.
> 
> The camera was hit or miss: it could detect some objects, and that's it if I remember correctly.

Thank you Siderate. I appreciat you replied after the long time. I dug out that v2.2.3_waveshare-esp32-s3-touch-lcd-3.5b.zip works for both 3.5B and 3.5C.