78/xiaozhi-esp32 #1076: 设备开机概率出现系统崩溃重启

liuxingxing1104, 2025-08-13:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.8.5_esp-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

系统正常开机

### What is the actual behavior?

系统崩溃重启

### Steps to reproduce.

没有特殊操作，就是给设备USB供电开机

### Debug Logs.

```plain
I (727) octal_psram: vendor id    : 0x0d (AP)
I (727) octal_psram: dev id       : 0x03 (generation 4)
I (727) octal_psram: density      : 0x05 (128 Mbit)
I (729) octal_psram: good-die     : 0x01 (Pass)
I (734) octal_psram: Latency      : 0x01 (Fixed)
I (738) octal_psram: VCC          : 0x00 (1.8V)
I (742) octal_psram: SRF          : 0x01 (Fast Refresh)
I (747) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (752) octal_psram: BurstLen     : 0x01 (32 Byte)
I (757) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (762) octal_psram: DriveStrength: 0x00 (1/1)
I (767) MSPI Timing: PSRAM timing tuning index: 6
I (770) esp_psram: Found 16MB PSRAM device
I (774) esp_psram: Speed: 80MHz
I (777) cpu_start: Multicore app
I (792) cpu_start: Pro cpu start user code
I (792) cpu_start: cpu freq: 240000000 Hz
I (792) app_init: Application information:
I (792) app_init: Project name:     xiaozhi
I (796) app_init: App version:      1.8.5
I (799) app_init: Compile time:     Aug 11 2025 11:19:12
I (804) app_init: ELF file SHA256:  5db4d4ae9...
I (809) app_init: ESP-IDF:          v5.4.1-dirty
I (813) efuse_init: Min chip rev:     v0.0
I (817) efuse_init: Max chip rev:     v0.99
I (821) efuse_init: Chip rev:         v0.2
I (825) heap_init: Initializing. RAM available for dynamic allocation:
I (831) heap_init: At 3FCB37A0 len 00035F70 (215 KiB): RAM
I (836) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
[... 173 lines cut ...]
--- 0x4200ab21: AudioService::Start()::{lambda(void*)#2}::_FUN(void*) at F:/esp_idf_workspace/xiaozhi/xiaozhi-esp32/main/audio/audio_service.cc:111
---  (inlined by) _FUN at F:/esp_idf_workspace/xiaozhi/xiaozhi-esp32/main/audio/audio_service.cc:113

ELF file SHA256: 5db4d4ae9

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379e15
--- 0x40379e15: esp_restart_noos at E:/Esp32/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x267c
load:0x403c8700,len:0x4
load:0x403c8704,len:0x117c
load:0x403cb700,len:0x3c2c
entry 0x403c8a30
I (23) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (23) boot: compile time Aug  7 2025 14:33:39
I (23) boot: Multicore bootloader
D (24) bootloader_flash: non-XMC chip detected by SFDP Read (C8), skip.
D (30) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
```

liuxingxing1104, 2025-08-13:
> 该问题已请教过虾哥，新版本已修复

您好，新版是指1.8.7吗？

Levi-Ann, 2025-08-13:
> > 该问题已请教过虾哥，新版本已修复
> 
> 您好，新版是指1.8.7吗？

是的

closed 2025-08-16 by 78