78/xiaozhi-esp32 #1093: 开机之后一直保存，检查新版本失败，无法进行对话

ljsaaa123, 2025-08-16:
E (93911) EspTcp: Failed to connect to 43.139.6.120:8089
E (93911) HttpClient: TCP connection failed
E (93911) Ota: Failed to open HTTP connection
W (93911) Application: Alert 错误: 检查新版本失败，将在 40 秒后重试：http://43.139.6.120:8089/lychee/ota/ [sad]
I (93931) AudioService: OpusHead: version=1, channels=1, sample_rate=16000
E (93931) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (93931) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
W (93931) Application: Check new version failed, retry in 40 seconds (3/10)
I (93941) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:24000 mask:1
I (93971) Adev_Codec: Open codec device OK
I (93971) AudioCodec: Set output enable to true
I (97971) SystemInfo: free sram: 76923 minimal sram: 73103
I (107971) SystemInfo: free sram: 76883 minimal sram: 73103
I (109931) AudioCodec: Set output enable to false
I (117971) SystemInfo: free sram: 76883 minimal sram: 73103
I (127971) SystemInfo: free sram: 76951 minimal sram: 73103
I (133941) Ota: Current version: 1.8.8
I (137971) SystemInfo: free sram: 73983 minimal sram: 73103
I (147971) SystemInfo: free sram: 73983 minimal sram: 73103
E (152441) EspTcp: Failed to connect to 43.139.6.120:8089
E (152441) HttpClient: TCP connection failed
E (152441) Ota: Failed to open HTTP connection
W (152441) Application: Alert 错误: 检查新版本失败，将在 80 秒后重试：http://43.139.6.120:8089/lychee/ota/ [sad]
I (152461) AudioService: OpusHead: version=1, channels=1, sample_rate=16000
E (152461) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (152461) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (152471) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:24000 mask:1
W (152461) Application: Check new version failed, retry in 80 seconds (4/10)

78, 2025-08-16:
看起来是你的服务器无法访问，刷一下官方服务器固件试试

f1jiang, 2025-08-16:
同样的问题，检查新版本失败，mark!