78/xiaozhi-esp32 #1094: Abortspeaking和listening的状态切换，无法在进入listening之后加入播放提醒音

NicLam26, 2025-08-16:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.7.6 xmini-c3-board

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

我希望在代码里面，在每次进入聆听模式之前，都播放popup声音，来提醒用户已经进入聆听模式，可以开始说话。

我的做法是：在setdevicestate()里面，case kDeviceStateListening的分支最后面，加入播放popup的声音。

[image]

[image]

### What is the actual behavior?

现象如下：
1. 上电-->进入idle-->按BOOT进入listening, 可以听到popup的声音，此时按BOOT再进idle, 再按BOOT进listening，也可以听到popup的声音
--说明idle和listening之间的切换，是正常想要的效果

2. 上电-->进入idle-->按BOOT进入listening, 可以听到popup的声音，此时用户说话，然后大模型说话，进入speaking, 大模型结束说话，进入listening, 也可以听到popup的声音
--说明speaking(正常结束)和listening之间的切换，是正常想要的效果

3. 上电-->进入idle-->按BOOT进入listening, 可以听到popup的声音，此时用户说话，然后大模型说话，进入speaking, 在大模型说话还没有结束的时候，按BOOT打断，abort speaking, 进入listening, 则听不到popup的声音
-说明abort speaking(打断speaking)和listening之间的切换，没有实现想要的结果

我尝试过很多种其他的方法，比如在AbortSpeaking()函数之前或者之后加播放声音，都没有效果。

请问现象3是为什么呢？是否bug? 有没有建议的修改方案？

### Steps to reproduce.

如上

### Debug Logs.

```plain
无
```

78, 2025-08-16:
可以更新到最新的代码，1.8.x对音频部分进行了重构，再进行测试。

NicLam26, 2025-08-17:
换成1.8.8可以了，我在进入聆听的最后加入播放popup，所有场景都可以了，看起来是1.7.6的问题。

1.8.8音频部分的重构是修改什么可以fix这个问题? （只是为了加深理解）

78, 2025-08-17:
> 换成1.8.8可以了，我在进入聆听的最后加入播放popup，所有场景都可以了，看起来是1.7.6的问题。
> 
> 1.8.8音频部分的重构是修改什么可以fix这个问题? （只是为了加深理解）

恐怕需要问一下 AI