78/xiaozhi-esp32 #1096: 1.8.5版本，之前的版本存在这个问题，使用的是Websocket的WSS通讯，经常出现异常重启

JasonChar, 2025-08-17:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.8.6

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

每次进入对话，有概率在对话过程中突然重启，没有发现规律

### What is the actual behavior?

报错日志如下：
I (476464) SystemInfo: free sram: 96959 minimal sram: 64579

assert failed: tlsf_free tlsf.c:629 (!block_is_free(block) && "block already marked as free")

Backtrace: 0x40379e15:0x3fce4960 0x40383411:0x3fce4980 0x4038a139:0x3fce49a0 0x40388faf:0x3fce4ac0 0x40388e6a:0x3fce4ae0 0x4037a5d3:0x3fce4b00 0x42141741:0x3fce4b20 0x4213f8e2:0x3fce4b40 0x42138536:0x3fce4b60 0x42138699:0x3fce4b80 0x42135eee:0x3fce4bc0 0x4213114e:0x3fce4be0 0x421c315e:0x3fce4c00 0x4201ecc3:0x3fce4c20 0x4201fd85:0x3fce4c50 0x42020be6:0x3fce4c90 0x42020c1d:0x3fce4d10 0x4201ef12:0x3fce4d30 0x4201ef1d:0x3fce4d70
--- 0x40379e15: panic_abort at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/esp_system/panic.c:454
--- 0x40383411: esp_system_abort at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/esp_system/port/esp_system_chip.c:87
--- 0x4038a139: __assert_func at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/newlib/assert.c:80
--- 0x40388faf: block_next at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/tlsf/tlsf_block_functions.h:161
---  (inlined by) block_link_next at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/tlsf/tlsf_block_functions.h:168
---  (inlined by) block_mark_as_free at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/tlsf/tlsf_block_functions.h:176
---  (inlined by) tlsf_free at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/tlsf/tlsf.c:630
--- 0x40388e6a: multi_heap_free_impl at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/multi_heap.c:233
---  (inlined by) multi_heap_free_impl at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/multi_heap.c:222
--- 0x4037a5d3: heap_caps_free at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/heap/heap_caps_base.c:75
--- 0x42141741: esp_mbedtls_mem_free at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/mbedtls/port/esp_mem.c:38
--- 0x4213f8e2: mbedtls_free at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/mbedtls/mbedtls/library/platform.c:54
--- 0x42138536: esp_mbedtls_free_buf at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/mbedtls/port/dynamic/esp_mbedtls_dynamic_impl.c:39
--- 0x42138699: esp_mbedtls_free_tx_buffer at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/mbedtls/port/dynamic/esp_mbedtls_dynamic_impl.c:306
--- 0x42135eee: __wrap_mbedtls_ssl_write at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/mbedtls/port/dynamic/esp_ssl_tls.c:327
--- 0x4213114e: esp_mbedtls_write at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/esp-tls/esp_tls_mbedtls.c:298
--- 0x421c315e: esp_tls_conn_write at D:/Espressif5.4.1/frameworks/esp-idf-v5.4.1/components/esp-tls/esp_tls.c:146
--- 0x4201ecc3: EspSsl::Send(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/esp/esp_ssl.cc:92
--- 0x4201fd85: WebSocket::SendControlFrame(unsigned char, void const*, unsigned int) at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/web_socket.cc:435
--- 0x42020be6: WebSocket::OnTcpData(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/web_socket.cc:385
--- 0x42020c1d: std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), WebSocket::Connect(char const*)::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/web_socket.cc:159
---  (inlined by) __invoke_impl<void, WebSocket::Connect(char const*)::<lambda(const std::string&)>&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> at D:/Espressif5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:61
---  (inlined by) __invoke_r<void, WebSocket::Connect(char const*)::<lambda(const std::string&)>&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> at D:/Espressif5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:111
---  (inlined by) _M_invoke at D:/Espressif5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:290
--- 0x4201ef12: std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const at D:/Espressif5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:591
---  (inlined by) EspSsl::ReceiveTask() at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/esp/esp_ssl.cc:133
--- 0x4201ef1d: EspSsl::Connect(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)::{lambda(void*)#1}::_FUN(void*) at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/esp/esp_ssl.cc:50
---  (inlined by) _FUN at D:/ESP32-S3/esp32-78-all/esp32-1.8.6-8.11/managed_components/78__esp-ml307/src/esp/esp_ssl.cc:53

ELF file SHA256: e7ae2f7bc

Rebooting...
I (29) octal_psram: vendor id    : 0x0d (AP)
I (29) octal_psram: dev id       : 0x02 (generation 3)
I (30) octal_psram: density      : 0x03 (64 Mbit)
I (31) octal_psram: good-die     : 0x01 (Pass)
I (35) octal_psram: Latency      : 0x01 
配置已经使用了：
# Fix ESP_SSL error
CONFIG_MBEDTLS_SSL_RENEGOTIATION=n

[image]

### Steps to reproduce.

使用Websocket通信，wss，有概率出现该异常，导致设备重启，帮忙分析一下最可能是什么原因呢？困扰很长时间了，感激不尽

### Build or installation Logs.

```plain

```

78, 2025-08-17:
wss 的 server 开了 ping？看起来像是回复server的 ping 的时候，跟你的设备正在发送数据产生了多线程资源竞争，可以考虑在 Send 的地方增加 mutex lock，或者关闭 server 的 keep alive。

JasonChar, 2025-08-19:
> wss 的 server 开了 ping？看起来像是回复server的 ping 的时候，跟你的设备正在发送数据产生了多线程资源竞争，可以考虑在 Send 的地方增加 mutex lock，或者关闭 server 的 keep alive。

感谢虾哥百忙之中的回复，是的，之前开了ping，但是把ping关闭之后，还是会有这个问题

[image]

JasonChar, 2025-09-06:
虾哥威武