78/xiaozhi-esp32 #1126: 可以增加一个播放ogg字节流的方法吗？

AllofLife, 2025-08-25:
### Is your feature request related to a problem?

我注意到1.8.8版本已经使用ogg文件进行播放提示音，AudioService::PlaySound 可以读取文件，播放数据，但是这个ogg解析不够完善 不能解析提取网络下载的ogg格式数据，虾哥有意向添加一个可以解析数据流的ogg函数吗？
这样方便下载音频数据直接播放，不需要再进行mp3解码。
我尝试修改一个版本，目前还没有成功。

78, 2025-08-25:
ogg只是一个容器，用来装编码后的音频流。你下载的ogg可能不是opus编码的，可能是vorbis编码，本项目里没有vorbis解码器。

78, 2025-08-30:
audio service除了3个tasks占用的堆栈内存，动态内存的消耗主要来自数据包中的队列，可以参考队列大小的定义：

```c
#define OPUS_FRAME_DURATION_MS 60
#define MAX_ENCODE_TASKS_IN_QUEUE 2
#define MAX_PLAYBACK_TASKS_IN_QUEUE 2
#define MAX_DECODE_PACKETS_IN_QUEUE (2400 / OPUS_FRAME_DURATION_MS)
#define MAX_SEND_PACKETS_IN_QUEUE (2400 / OPUS_FRAME_DURATION_MS)
```

可以计算 decode packets 和 playback tasks 都满载时，内存消耗大约是 opus包大小 * (2400/60) + pcm包大小 * 2，opus包大小通常为200字节以内，pcm包大小通常为2KB，远没有到你提及的 90KB。可以分析是否在其他地方有内存消耗，例如大量调用 Schedule 增加任务队列也会消耗内存。

AllofLife, 2025-09-02:
[image]
感谢虾哥的指导 目前已经实现了ogg网络流文件的播放，但是在调试中经常出现：设备听到我说的话，本来要进入说话中的时候，却没有说，立刻进入聆听中。想问下这个可能是哪里的bug呢？
I (391148) Application: >> 播放张杰的逆战。
I (391148) Application: STATE: speaking
I (391868) Application: STATE: listening
I (397648) Application: >> 播放梁静茹的暖暖。
I (397648) Application: STATE: speaking
I (398438) Application: STATE: listening
I (402428) wifi:Set ps type: 1, coexist: 0