78/xiaozhi-esp32 #1131: 小智1.8.5版本，使用ML307的tcp连接服务器发送语音时疑似有数据阻塞

yeyeyeah233, 2025-08-27:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.8.5

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

切换到聆听状态时，模块放着不动过一段时间就会触发这个，用udp连接小智的服务器就不会出现这种状况

### What is the actual behavior?

E (30498) Ml307Tcp: Failed to send data chunk

### Steps to reproduce.

用面包板+307模组就能复现

### Debug Logs.

```plain
I (29308) AtUart: << +MIPSEND: 1,7 (13 bytes)
I (29308) AtUart: << OK (2 bytes)
I (29368) AtUart: >> AT+MIPSEND=1,7,8281A859065CF0
 (31 bytes)
I (29368) AtUart: << +MIPSEND: 1,7 (13 bytes)
I (29368) AtUart: << OK (2 bytes)
I (29388) AtUart: << +MIPURC: "rtcp",1,6,8904B594B706 (32 bytes)
I (29388) AtUart: >> AT+MIPSEND=1,10,8A842DCF4629985BF12F
 (38 bytes)
I (29428) AtUart: >> AT+MIPSEND=1,7,8281C087FD0498
 (31 bytes)
E (29488) Ml307Tcp: Failed to send data chunk
I (29488) AtUart: >> AT+MIPCLOSE=1 (13 bytes)
I (30488) WS: Websocket disconnected
I (30488) AtUart: << +MIPSEND: 1,10 (14 bytes)
I (30488) AtUart: << OK (2 bytes)
I (30488) AtUart: << OK (2 bytes)
I (30488) AtUart: << +MIPCLOSE: 1 (12 bytes)
I (30488) AtUart: << +CME ERROR: 550 (15 bytes)
E (30498) Ml307Tcp: Failed to send data chunk
I (30498) Application: STATE: idle
I (31668) AtUart: >> AT+CSQ (6 bytes)
I (31668) AtUart: << +CSQ: 31,99 (11 bytes)
I (31668) AtUart: << OK (2 bytes)
```

closed 2025-08-27 by yeyeyeah233

yeyeyeah233, 2025-08-28:
> 尝试以下两个方案：
> 
> 1. 升级到 esp-ml307 最新版本 3.3.0 https://github.com/78/esp-ml307
> 2. 如果方案1未解决，尝试在 ML307 的 Send 方法里加线程锁

我已经升级到了3.3.0的版本，还是出现一样的状况，而且3.3.0SendCommandWithData函数里好像已经加过线程锁了

yeyeyeah233, 2025-09-01:
> > > 尝试以下两个方案：
> > > 
> > > 1. 升级到 esp-ml307 最新版本 3.3.0 https://github.com/78/esp-ml307
> > > 2. 如果方案1未解决，尝试在 ML307 的 Send 方法里加线程锁
> > 
> > 
> > 我已经升级到了3.3.0的版本，还是出现一样的状况，而且3.3.0SendCommandWithData函数里好像已经加过线程锁了
> 
> 是在外面的那层加

请问具体是在哪些方法里加

yeyeyeah233, 2025-09-01:
> 从debug的log来看，是在
> 
> ```
> I (29388) AtUart: << +MIPURC: "rtcp",1,6,8904B594B706 (32 bytes)
> ```
> 
> 之后出现故障，这个是服务器发送的 ping，之前有网友反馈，关闭服务器的 websocket 的 ping 后，问题已经得到解决。

但是1.6.1的版本的log也收到了同样的ping，程序没卡住

78, 2025-09-02:
已修复 websocket ping 导致连接断开的问题：https://github.com/78/esp-ml307/commit/c2898e4f415bbd14fc223005852db39e52b25eec