78/xiaozhi-esp32 #1142: MCP工具调用偶发调用异常

Ycmg, 2025-08-28:
大模型回应已经调用完工具了，但实际并没有调用到实际的工具

Ycmg, 2025-08-28:
> 清除智能体的记忆再试试，会正常吗？

好的，我试试

Ycmg, 2025-08-28:
> 清除智能体的记忆再试试，会正常吗？

不正常，还会偶发没有调用工具，但大模型已经回应完成了

78, 2025-08-28:
你说的这个情况，通常也由于提示词存在干扰导致

JasonChar, 2025-08-30:
我也遇到了，死活不调用工具，有时候又可以

CoreyChang, 2025-09-01:
@78 我发现单次会话，如果对话数多了，角色介绍和MCP工具的描述，它们的权重就变低了。控制就变得不精准，甚至有可能就不调用MCP了，中间的会话我都是一直在控制，没有沟通其它的。

qq749812679, 2025-09-01:
你可以把调用mcp下发指令的日志也打印出来看看，先看看指令执行了吗

HellCatVN, 2025-10-31:
> > [@78](https://github.com/78) I have change "Memory Type" into "No Memory" seem like LLM response and hang up instead changing stay and called MCP tool . Unless I press "Boot" Button.
> > I (250164) Application: STATE: speaking I (250164) Application: >> Có thể phát cho tôi bài hát Bạn đời của Karit không? I (251224) Application: << Dạ được ạ, I (251234) AudioCodec: Set output enable to true I (252124) Application: << mình sẽ phát bài hát "Bạn đời" của Karit ngay đây I (255484) Application: << ạ. I (259524) SystemInfo: free sram: 102595 minimal sram: 100643 I (269524) SystemInfo: free sram: 102591 minimal sram: 100643 I (272234) AudioCodec: Set output enable to false I (279524) SystemInfo: free sram: 103111 minimal sram: 100643 I (289524) SystemInfo: free sram: 102591 minimal sram: 100643 I (299524) SystemInfo: free sram: 103111 minimal sram: 100643 I (309524) SystemInfo: free sram: 102595 minimal sram: 100643 I (315974) MQTT: Received goodbye message, session_id: 9f25949a I (315974) wifi:Set ps type: 1, coexist: 0
> > I (315974) Application: STATE: idle I (325524) SystemInfo: free sram: 107467 minimal sram: 100643
> 
> Try switching the LLM model to DeepSeek V3 and see if it gets better?

Changing LLM model doesn't help I have changed it and retry multiple times

78, 2025-10-31:
> > > > Try switching the LLM model to DeepSeek V3 and see if it gets better?
> > > 
> > > 
> > > Changing LLM model doesn't help I have changed it and retry multiple times
> > 
> > 
> > It seems this is a bug that only happens on some specific devices. What is your board type? (You can get it from xiaozhi.me -> Manage Devices)
> 
> I am using esp32 s3 n16r8 . from provided log and what I tried to debug highly percentage is server not calling mcp tools it not about device . Also a lot of people in our community facing this issue

If possible, you can send the complete reproduction steps for this bug and a full serial port monitoring log to my email: terrence@tenclass.com. My team and I will assist in analyzing the cause.

78, 2025-11-01:
> I am using esp32 s3 n16r8 . from provided log and what I tried to debug highly percentage is server not calling mcp tools it not about device . Also a lot of people in our community facing this issue

We have located the issue now. It might be fixed in 24 hours.

Aogu181, 2026-10-22:
> 21\. > > > > Try switching the LLM model to DeepSeek V3 and see if it gets better?试试把 LLM 模型切换到 DeepSeek V3，看看会不会变好？
>      > > > 
>      > > > 
>      > > > Changing LLM model doesn't help I have changed it and retry multiple times更改 LLM 模型也没用，我已经多次更改并重试过
>      > > 
>      > > 
>      > > It seems this is a bug that only happens on some specific devices. What is your board type? (You can get it from xiaozhi.me -> Manage Devices)看起来这是只在某些特定设备上出现的 bug。你的板子类型是什么？（你可以从 xiaozhi.me ->管理设备获取）
>      > 
>      > 
>      > I am using esp32 s3 n16r8 . from provided log and what I tried to debug highly percentage is server not calling mcp tools it not about device . Also a lot of people in our community facing this issue我用的是 esp32 s3 n16r8。根据提供的日志和我尝试调试的情况，很大比例是服务器没有调用 MCP 工具，这和设备无关。我们社区里也有很多人面临这个问题
>      
>      
>      We have located the issue now. It might be fixed in 24 hours.我们现在已经找到了问题所在。可能24小时内能修复。

有解决吗？
我最近也老是遇到有时候死活都无法调用
[image]

[image]

78, 2026-10-22:
@Aogu181 看起来不像是MCP的问题，可能tools太多没有上传完整，试试DeepSeek。为什么每次做完动作会自动退下？是故意的吗？还是说连续超过3小时使用被限制？等1小时后再试试看。

Aogu181, 2026-10-22:
> 23\. [@Aogu181](https://github.com/Aogu181) 看起来不像是MCP的问题，可能tools太多没有上传完整，试试DeepSeek。为什么每次做完动作会自动退下？是故意的吗？还是说连续超过3小时使用被限制？等1小时后再试试看。

它超时就自动关闭了，就聊了一句。我确实有很多MCP工具，怎么看有没有上传完整呢？我看服务器拉tools的时候好像都有发。还有3小时使用限制是什么

78, 2026-10-23:
> 24\. > 23. [@Aogu181](https://github.com/Aogu181) 看起来不像是MCP的问题，可能tools太多没有上传完整，试试DeepSeek。为什么每次做完动作会自动退下？是故意的吗？还是说连续超过3小时使用被限制？等1小时后再试试看。
>      
>      
>      它超时就自动关闭了，就聊了一句。我确实有很多MCP工具，怎么看有没有上传完整呢？我看服务器拉tools的时候好像都有发。还有3小时使用限制是什么

防沉迷。单次对话时长不超过3小时，单日累计时长不超过6小时。

Aogu181, 2026-10-27:
> 23\. [@Aogu181](https://github.com/Aogu181) 看起来不像是MCP的问题，可能tools太多没有上传完整，试试DeepSeek。为什么每次做完动作会自动退下？是故意的吗？还是说连续超过3小时使用被限制？等1小时后再试试看。

为什么有时候initialize 之后立刻就是 tools/list， 有时候没有tools/list 打印呢

Aogu181, 2026-10-27:
> 23\. [@Aogu181](https://github.com/Aogu181) 看起来不像是MCP的问题，可能tools太多没有上传完整，试试DeepSeek。为什么每次做完动作会自动退下？是故意的吗？还是说连续超过3小时使用被限制？等1小时后再试试看。

基于 xiaozhi-esp32 **v2.0.4**

**正常会话**（initialize 后 ~0.6s 即拉取，两页完整）：

```
I (13333) MCP: initialize id=10000
I (13403) MCP: tools/list id=10001 cursor='' user=0: sent 15
    [self.get_device_status, …, self.light.schedule_control, self.power.control,
     self.robot.action, self.robot.stop, self.battery.get_level] next='add_reminder'
I (13583) MCP: tools/list id=10002 cursor='add_reminder' user=0: sent 3
    [add_reminder, delete_reminder, list_reminders] next=''
```

**异常会话**（initialize 后再无任何 tools/list）：

```
I (13333) MCP: initialize id=10000
    ← 之后再无 tools/list；同一连接的音频/唤醒/对话正常
…
I (31783) Application: << 抱歉，我没有查看闹钟列表的工具哦。
I (86353) Application: << 抱歉，我没有查询IP地址的工具哦。
I (113984) Application: << 我发现当前可用的工具列表中没有前进相关的动作控制工具。
```

2026-10-27 同日采集：**异常 5 例 / 正常 3 例**，随机出现于相同固件、相同网络。

## 设备侧

1. **initialize 应答正常**：异常会话里同一 MQTT 连接的音频推流、唤醒、对话、
   服务器工具调用全部正常——连接与发布通路健康，是后端在 initialize 之后
   不发起后续请求；
2. **与固件版本无关**：对比上游 xiaozhi-esp32 v2.5.2 的 `main/mcp_server.cc`，
   initialize 应答形状、tools/list 分页、withUserTools 参数与 v2.0.4 逻辑
   一致，设备侧 MCP server 行为无差异；
3. **工具注册完整**：每次启动 25 个工具全部注册成功（`MCP: Add tool: …`
   日志逐条可见），列表内容与返回页面正常会话时完全一致；
4. **设备侧主动轻推已试、被证实不可用**：设备按 MCP 规范发送
   `notifications/tools/list_changed` 请求客户端重新拉取，后端在 **10ms 内**
   回 `goodbye`（**空 session_id**）拆除会话