78/xiaozhi-esp32 #1156: ESP32S3编译出现esp-sr缺少

Zohlz, 2025-09-04:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.8.9

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

之前版本编译esp32s3时，能正常运行

### What is the actual behavior?

这两天清除build后，再次编译发现缺少
cannot find v1.0/v1.0.1/xiaozhi-esp32-main/managed_components/espressif__esp-sr/esp-tts/esp_tts_chinese/esp32s3: No such file or directory
重新克隆过，也尝试过换idf版本，都没办法解决
[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
*  正在执行任务: d:\idf-5.5.1\path\tools\ninja\1.12.1\ninja.EXE  

[0/1] Re-running CMake...-- Minimal build - OFF
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
NOTICE: Skipping optional dependency: waveshare/esp_lcd_jd9365_10_1
NOTICE: Skipping optional dependency: waveshare/esp_lcd_st7703
NOTICE: Skipping optional dependency: espressif/esp_lcd_ili9881c
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espfriends/servo_dog_ctrl
NOTICE: Processing 46 dependencies:
NOTICE: [1/46] 78/esp-ml307 (3.3.1)
NOTICE: [2/46] 78/esp-opus (1.0.5)
NOTICE: [3/46] 78/esp-opus-encoder (2.4.1)
NOTICE: [4/46] 78/esp-wifi-connect (2.5.2)
NOTICE: [5/46] 78/esp_lcd_nv3023 (1.0.0)
NOTICE: [6/46] 78/xiaozhi-fonts (1.5.2)
NOTICE: [7/46] espressif/adc_battery_estimation (0.2.0)
NOTICE: [8/46] espressif/adc_mic (0.2.1)
NOTICE: [9/46] espressif/button (4.1.3)
NOTICE: [10/46] espressif/cmake_utilities (0.5.3)
NOTICE: [11/46] espressif/dl_fft (0.3.1)
[... 287 lines cut ...]
[122/123] Generating binary image from built executable
esptool.py v4.10.dev2
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/build/bootloader/bootloader.bin
[123/123] C:\WINDOWS\system32\cmd.exe /C "cd /D "E:\xiaozhi\ZZPET-BOT-LCD v1.0\v1.0.1\xiaozhi-esp32-main\build\bootloader\esp-idf\esptool_py" && d:\idf-5.5.1\path\python_env\idf5.5_py3.11_env\Scripts\python.exe D:/idf-5.5.1/v5.5.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 "E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/build/bootloader/bootloader.bin""
Bootloader binary size 0x4050 bytes. 0x3fb0 bytes (50%) free.
[2247/2252] Generating esp-idf/esp_system/ld/sections.ld
warning: the default selection LCD_ST7789 (defined at E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:434, E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:448) of <choice DISPLAY_ESP32S3_KORVO2_V3> (defined at E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:428) is not contained in the choice
warning: the choice symbol LCD_ST7789 (defined at E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:434, E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:448) is defined with a prompt outside the choice
warning: the choice symbol LCD_ST7789 (defined at E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:434, E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/main/Kconfig.projbuild:448) is defined with a prompt outside the choice
info: INFO: Symbol LCD_ST7789 defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    E:\xiaozhi\ZZPET-BOT-LCD v1.0\v1.0.1\xiaozhi-esp32-main\main\Kconfig.projbuild:448
    E:\xiaozhi\ZZPET-BOT-LCD v1.0\v1.0.1\xiaozhi-esp32-main\main\Kconfig.projbuild:434
[2249/2252] Linking CXX executable xiaozhi.elf
FAILED: xiaozhi.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && D:\idf-5.5.1\path\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -nostartfiles    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 "-Wl,--Map=E:/xiaozhi/ZZPET-BOT-LCD v1.0/v1.0.1/xiaozhi-esp32-main/build/xiaozhi.map" -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.bt_funcs.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.ble_master.ld -T esp32s3.rom.ble_50.ld -T esp32s3.rom.ble_smp.ld -T esp32s3.rom.ble_dtm.ld -T esp32s3.rom.ble_test.ld -T esp32s3.rom.ble_scan.ld -T esp32s3.rom.libc.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\xiaozhi.elf.rsp -o xiaozhi.elf && cd ."
D:/idf-5.5.1/path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find v1.0/v1.0.1/xiaozhi-esp32-main/managed_components/espressif__esp-sr/lib/esp32s3: No such file or directory
D:/idf-5.5.1/path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find v1.0/v1.0.1/xiaozhi-esp32-main/managed_components/espressif__esp-sr/esp-tts/esp_tts_chinese/esp32s3: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  终端进程“d:\idf-5.5.1\path\tools\ninja\1.12.1\ninja.EXE”已终止，退出代码: 1。
```

78, 2025-09-04:
Consider switching another network?

Zohlz, 2025-09-05:
> Consider switching another network?

谢谢大佬，我找到问题了，是我的文件夹命名带了空格导致的