78/xiaozhi-esp32 #1206: LvglGif: Failed to open GIF from image descriptor

hecai84, 2025-09-16:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

2.01

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

给小智设置一个gif表情

### What is the actual behavior?

我用2.0.1的代码编译的,然后用小智 AI 自定义资源生成器,设置了一个Gif表情.
重启了之后,gif显示不出来,看日志报错.
gif格式是GIF89a.
是哪个地方操作有问题吗?还是说gif的尺寸大小有限制呢.

E (18965) LvglGif: Failed to open GIF from image descriptor
E (18975) LcdDisplay: Failed to load GIF for emotion: neutral

### Steps to reproduce.

1.编译
2.烧录
3.在线设置表情
4.下载asset.bin

### Debug Logs.

```plain

```

### More Information.

E (18965) LvglGif: Failed to open GIF from image descriptor
E (18975) LcdDisplay: Failed to load GIF for emotion: neutral

hecai84, 2025-09-16:
设置png图片是没问题的.

hecai84, 2025-09-16:
> 重新拉一下main的代码，应该能打印一下Failed to open GIF from image descriptor的原因

之前放不了gif,用ps 给它在颜色表里面选个颜色就可以用了.
photoshop->文件->导出->存储为web所用格式->颜色表->存储

78, 2025-09-16:
> > 重新拉一下main的代码，应该能打印一下Failed to open GIF from image descriptor的原因
> 
> 之前放不了gif,用ps 给它在颜色表里面选个颜色就可以用了. photoshop->文件->导出->存储为web所用格式->颜色表->存储

GIF播放代码来自LVGL项目，看来是不支持没有color table的情况