78/xiaozhi-esp32 #1215: 添加远程唤醒MCP,能唤醒有文字返回，就是没有声音

1cccux, 2025-09-17:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v2.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

我喜欢远程唤醒并播放回复音

### What is the actual behavior?

无法播放唤醒回复音

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

### More Information.

虾哥你好，我尝试添加了用户自定义MCP，想通过MQTT下发指令远程自定义文字唤醒设备，但是发现服务器有返回文字，首段音频没办法播放出来，如果有空，帮忙看看代码怎么改
```
 AddUserOnlyTool("self.wakeup", "Wake up module with parameters",
    PropertyList({
        Property("reason", kPropertyTypeString, "Reason for waking up"),
    }),
    [this](const PropertyList& properties) -> ReturnValue {

        
        std::string reason = "你好小智";
        try {
            reason = properties["reason"].value<std::string>();
        }
        catch (...) {
        }
        ESP_LOGI(TAG, "Wakeup MCP: reason=%s", reason.c_str());

        auto& app = Application::GetInstance();
        // 先打开音频通道
        if (app.GetProtocol() && !app.GetProtocol()->IsAudioChannelOpened()) {
            app.GetProtocol()->OpenAudioChannel();
        }

        auto display = Board::GetInstance().GetDisplay();
        if (display) {
            display->SetEmotion("happy");
            display->SetStatus(reason.c_str());
            display->ShowNotification(reason.c_str(), 4000);
            display->SetChatMessage("system", reason.c_str());
        }
        if (app.GetDeviceState() == kDeviceStateIdle) {
            app.Schedule([&app, reason]() mutable {
                if (!app.GetProtocol()) {
                    return;
                }
                // 打开音频通道
                if (!app.GetProtocol()->IsAudioChannelOpened()) {
                    if (!app.GetProtocol()->OpenAudioChannel()) {
                        return;
                    }
                }
                app.GetProtocol()->SendWakeWordDetected(reason);
                app.GetAudioService().EnableWakeWordDetection(false);
                esp_timer_handle_t delayed_listen_timer;
                esp_timer_create_args_t timer_args = {
                    .callback = [](void* arg) {
                        Application* app = static_cast<Application*>(arg);
                        app->Schedule([app]() {
                            if (app->GetDeviceState() == kDeviceStateIdle) {
                                app->StartListening();
                            }
                        });
                    },
                    .arg = &app,
                    .name = "delayed_listen_timer"
                };
                esp_timer_create(&timer_args, &delayed_listen_timer);
                esp_timer_start_once(delayed_listen_timer, 1000000); 
            });
        }
     return true;
 }
```

78, 2025-09-17:
目前仅支持语音唤醒，不支持文字唤醒。即首句话必须是user发起的语音。

openrz, 2025-09-19:
[悲伤]虾哥不用xiaozhi-mqtt-gateway了？

hackers365, 2025-09-19:
> > 如果单纯的文字唤醒无法实现。 我看闹钟是已经实现了：倒计时文本唤醒的。那么是不是可以通过：设置倒计时为1秒的闹钟，间接实现文字唤醒？
> 
> 对于发起对话，Websocket是没有太多限制的。目前MQTT+UDP协议规定了，需要设备端先发送语音，语音通道才能成功建立，否则服务器无法知道设备端的UDP地址。

现在设备启动初始化是没有 建立mqtt连接的，能否优化为mqtt保持长连接不断开，这样就可以有信令通道下发命令了

78, 2025-09-19:
> > > 如果单纯的文字唤醒无法实现。 我看闹钟是已经实现了：倒计时文本唤醒的。那么是不是可以通过：设置倒计时为1秒的闹钟，间接实现文字唤醒？
> > 
> > 
> > 对于发起对话，Websocket是没有太多限制的。目前MQTT+UDP协议规定了，需要设备端先发送语音，语音通道才能成功建立，否则服务器无法知道设备端的UDP地址。
> 
> 现在设备启动初始化是没有 建立mqtt连接的，能否优化为mqtt保持长连接不断开，这样就可以有信令通道下发命令了

只要 OTA 接口返回了 MQTT 接入点，就默认会选择 MQTT 进行连接。你的情况应该是 server 没有支持。

1cccux, 2025-09-25:
> > > > 如果单纯的文字唤醒无法实现。 我看闹钟是已经实现了：倒计时文本唤醒的。那么是不是可以通过：设置倒计时为1秒的闹钟，间接实现文字唤醒？
> > > 
> > > 
> > > 对于发起对话，Websocket是没有太多限制的。目前MQTT+UDP协议规定了，需要设备端先发送语音，语音通道才能成功建立，否则服务器无法知道设备端的UDP地址。
> > 
> > 
> > 现在设备启动初始化是没有 建立mqtt连接的，能否优化为mqtt保持长连接不断开，这样就可以有信令通道下发命令了
> 
> 只要 OTA 接口返回了 MQTT 接入点，就默认会选择 MQTT 进行连接。你的情况应该是 server 没有支持。

这是我修改后的代码，声音有时有有时没有，有点随机性，没有声音都会出现如下日志，是不是哪里冲突了，如果可以解决该怎么解决呢
AddUserOnlyTool("self.wakeup", "Wake up module with parameters",
    PropertyList({
        Property("reason", kPropertyTypeString, "Reason for waking up"),
    }),
    [this](const PropertyList& properties) -> ReturnValue {
        std::string reason = "你好小智";
        try {
            reason = properties["reason"].value<std::string>();
        }
        catch (...) {
        }
        ESP_LOGI(TAG, "Wakeup MCP: reason=%s", reason.c_str());
        auto& app = Application::GetInstance();
        auto display = Board::GetInstance().GetDisplay();
        if (display) {
            display->SetEmotion("happy");
        }
       app.Schedule([reason]() {
        auto& app = Application::GetInstance();
        if (reason.empty()) {
            app.WakeWordInvoke("你好小智");
        } else {
            app.WakeWordInvoke(reason);
        }
    });
   
        return true;
    }
);

[image]

78, 2025-09-25:
> > > > > 如果单纯的文字唤醒无法实现。 我看闹钟是已经实现了：倒计时文本唤醒的。那么是不是可以通过：设置倒计时为1秒的闹钟，间接实现文字唤醒？
> > > > 
> > > > 
> > > > 对于发起对话，Websocket是没有太多限制的。目前MQTT+UDP协议规定了，需要设备端先发送语音，语音通道才能成功建立，否则服务器无法知道设备端的UDP地址。
> > > 
> > > 
> > > 现在设备启动初始化是没有 建立mqtt连接的，能否优化为mqtt保持长连接不断开，这样就可以有信令通道下发命令了
> > 
> > 
> > 只要 OTA 接口返回了 MQTT 接入点，就默认会选择 MQTT 进行连接。你的情况应该是 server 没有支持。
> 
> 这是我修改后的代码，声音有时有有时没有，有点随机性，没有声音都会出现如下日志，是不是哪里冲突了，如果可以解决该怎么解决呢 AddUserOnlyTool("self.wakeup", "Wake up module with parameters", PropertyList({ Property("reason", kPropertyTypeString, "Reason for waking up"), }), [this](const PropertyList& properties) -> ReturnValue { std::string reason = "你好小智"; try { reason = properties["reason"].valuestd::string(); } catch (...) { } ESP_LOGI(TAG, "Wakeup MCP: reason=%s", reason.c_str()); auto& app = Application::GetInstance(); auto display = Board::GetInstance().GetDisplay(); if (display) { display->SetEmotion("happy"); } app.Schedule(reason { auto& app = Application::GetInstance(); if (reason.empty()) { app.WakeWordInvoke("你好小智"); } else { app.WakeWordInvoke(reason); } });
> 
> ```
>     return true;
> }
> ```
> 
> );
> 
> <img alt="Image" width="817" height="325" src="https://private-user-images.githubusercontent.com/230930087/493642894-f62bf423-dd02-4f0f-b65f-415ab3661cc9.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg4MzQwMzAsIm5iZiI6MTc1ODgzMzczMCwicGF0aCI6Ii8yMzA5MzAwODcvNDkzNjQyODk0LWY2MmJmNDIzLWRkMDItNGYwZi1iNjVmLTQxNWFiMzY2MWNjOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyNVQyMDU1MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMjVmYmZlM2FhZGYwZTRiMjhlZmQ3MTE2MTQ4MzQ1Nzk4MmM2Mzk1OWU5NzlhOGJiMmVlM2JhYTU2NDE3MDUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hFbJj2bH0DygRJX0tsdwjf9jeWOKVgVOuIY7uW7jMCY">

前面已经说过原因了，这个问题继续讨论没有意义。

> 目前仅支持语音唤醒，不支持文字唤醒。即首句话必须是user发起的语音。

closed 2025-09-25 by 78