78/xiaozhi-esp32 #1220: ESP32P4配网后反复重启

konglingboy, 2025-09-18:
v2.0.2_waveshare-p4-wifi6-touch-lcd-4b，刷写后显示和语音正常，配网后正常连接，进行到检查新版本时crash,重启，跟踪日志：
`I (10803) Ota: Current version: 2.0.1

[08:38:41.872]收←◆E (11355) FreeRTOS: Fatal error: TLSP deletion callback at index 0 overwritten with non-excutable pointer 0x48007288

[08:38:41.923]收←◆
abort() was called at PC 0x4ff0b1f5 on core 1
Core  1 register dump:
MEPC    : 0x4ff00c1a  RA      : 0x4ff09b02  SP      : 0x4ff300a0  GP      : 0x4ff13a80  
TP      : 0x4ff30190  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4ff300dc  S1      : 0x4ff300dc  A0      : 0x4ff300dc  A1      : 0x4ff300be  
A2      : 0x00000000  A3      : 0x4ff30109  A4      : 0x00000001  A5      : 0x4ff1f000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4ff300c0  S3      : 0x4ff19000  
S4      : 0x4ff1f000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000001  

Stack memory:
4ff300a0: 0x00000001 0x4ff3010c 0x4ff300dc 0x4ff11ed8 0x00000000 0x00000000 0x00000000 0x00000031
4ff300c0: 0x30666634 0x35663162 0x4ff1f000 0x4ff180c8 0x4ff300c0 0x4ff180e4 0x4ff300bc 0x726f6261
4ff300e0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x62306666 0x20356631
4ff30100: 0x63206e6f 0x2065726f 0x00000031 0x00000000 0x00000000 0x4ff1f000 0x4ff6692c 0x4ff0b1f8
4ff30120: 0x4ff1f000 0x00000000 0x4ff1f000 0x4ff19000 0x00000000 0x4ff1f000 0x4ff6692c 0x4ff0c76c
4ff30140: 0x00000000 0x4ff1f000 0x4ff6692c 0x4ff0c85a 0x00000000 0x00000000 0x00000000 0x00000000
4ff30160: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff30180: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x4ff2fbb4 0x4ff2fbb4 0x00000000
4ff301a0: 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x4ff300d0 0x00000000
4ff301c0: 0x4ff1910c 0x4ff2fa54 0x4ff301b8 0x4ff19104 0x00000019 0x00000000 0x00000000 0x4ff301b8
4ff301e0: 0x00000000 0x00000000 0x4ff2fbb4 0x454c4449 0x00000031 0x00000000 0x00000000 0x00000001
4ff30200: 0x4ff301b0 0x0000000c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00a3b86f
4ff30220: 0x00000000 0x4ff2046c 0x4ff204d4 0x4ff2053c 0x00000000 0x00000000 0x00000001 0x00000000
4ff30240: 0x00000000 0x00000000 0x48006118 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff30260: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff30280: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff302a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff302c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff302e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff30300: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000200 0x00000800 0xa5a5a5a5
4ff30320: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30340: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30360: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30380: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff303a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff303c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff303e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30400: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30420: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30440: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30460: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff30480: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 1d37d20ac

Rebooting...`

nervYu, 2025-09-18:
这个是 esp-idf 版本的问题 需要升级到 5.5 可以解决

closed 2025-09-19 by konglingboy

HonestQiao, 2025-09-26:
今天测试了，确实5.5，能够正常运行。