78/xiaozhi-esp32 #1223: Esp32s3 N8R8 repeat boot

Xuantam91, 2025-09-21:
I am using a device with chip esp32s3 N8R8, original fw is 1.60 can run well, but display chinese, wake word is chinese also. I try to update latest 2.0.2. Config flash 8M, but it is be repeating boot after upgrading.  can you help to solve this?

78, 2025-09-21:
> I try many time, now bootloop issue isn't happening but wake word isn't not work. And can't not talk to AI ( like MIC is not working). But if reflash fw_1.6.bin it is working well. This is product infor. Can you advice which fw is work with this? Esp32s3 N8R8, Oled 0.96 128×64
> 
> https://gcnnjyl5fvdz.feishu.cn/wiki/CLODwXvviiXbHlkvHPjc1vpIn8d?from=from_parent_docs
> 
> https://rcn4qyi58ici.feishu.cn/wiki/CepQwJWkbiZ2nTkdqu5cVCD6ng3

You can upload your sdkconfig file in your project root. Then I can tell if you have configured it correctly.

Xuantam91, 2025-09-21:
thank you, this is my sdkconfig file ver 1.9.2 & below is log after test

[sdkconfig.txt](https://github.com/user-attachments/files/22452547/sdkconfig.txt)
`export IDF_PATH='/Users/baoan/esp/v5.5.1/esp-idf'
'/Users/baoan/.espressif/python_env/idf5.5_py3.9_env/bin/python' '/Users/baoan/esp/v5.5.1/esp-idf/tools/idf_monitor.py' -p /dev/tty.usbmodem142201 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --make ''/Users/baoan/.espressif/python_env/idf5.5_py3.9_env/bin/python' '/Users/baoan/esp/v5.5.1/esp-idf/tools/idf.py'' --target esp32s3 '/Users/baoan/Documents/GLOBY/Software/xiaozhi-esp32-1.9.2/build/xiaozhi.elf'
baoan@MBP-cua-BAO xiaozhi-esp32-1.9.2 % export IDF_PATH='/Users/baoan/esp/v5.5.1/esp-idf'
baoan@MBP-cua-BAO xiaozhi-esp32-1.9.2 % '/Users/baoan/.espressif/python_env/idf5.5_py3.9_env/bin/python' '/Users/baoan/es
p/v5.5.1/esp-idf/tools/idf_monitor.py' -p /dev/tty.usbmodem142201 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --make
 ''/Users/baoan/.espressif/python_env/idf5.5_py3.9_env/bin/python' '/Users/baoan/esp/v5.5.1/esp-idf/tools/idf.py'' --targ
et esp32s3 '/Users/baoan/Documents/GLOBY/Software/xiaozhi-esp32-1.9.2/build/xiaozhi.elf'
--- Warning: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbmodem142201 instead...
--- esp-idf-monitor 1.8.0 on /dev/cu.usbmodem142201 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037ef7e
--- 0x4037ef7e: esp_cpu_wait_for_intr at /Users/baoan/esp/v5.5.1/esp-idf/components/esp_hw_support/cpu.c:64
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x548
load:0x403c8700,len:0xc3c
load:0x403cb700,len:0x2e70
entry 0x403c8908
I (33) octal_psram: vendor id    : 0x0d (AP)
I (33) octal_psram: dev id       : 0x02 (generation 3)
I (33) octal_psram: density      : 0x03 (64 Mbit)
I (35) octal_psram: good-die     : 0x01 (Pass)
I (39) octal_psram: Latency      : 0x01 (Fixed)
I (44) octal_psram: VCC          : 0x01 (3V)
I (48) octal_psram: SRF          : 0x01 (Fast Refresh)
I (52) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (57) octal_psram: BurstLen     : 0x01 (32 Byte)
I (62) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (67) octal_psram: DriveStrength: 0x00 (1/1)
I (71) MSPI Timing: PSRAM timing tuning index: 5
I (75) esp_psram: Found 8MB PSRAM device
I (79) esp_psram: Speed: 80MHz
I (82) cpu_start: Multicore app
I (96) cpu_start: Pro cpu start user code
I (96) cpu_start: cpu freq: 240000000 Hz
I (96) app_init: Application information:
I (97) app_init: Project name:     xiaozhi
I (100) app_init: App version:      1.9.2
I (104) app_init: Compile time:     Sep 21 2025 22:01:10
I (109) app_init: ELF file SHA256:  93015101a...
I (113) app_init: ESP-IDF:          v5.5.1-dirty
I (117) efuse_init: Min chip rev:     v0.0
I (121) efuse_init: Max chip rev:     v0.99 
I (125) efuse_init: Chip rev:         v0.2
I (129) heap_init: Initializing. RAM available for dynamic allocation:
I (135) heap_init: At 3FCA98F0 len 0003FE20 (255 KiB): RAM
I (140) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (146) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (151) heap_init: At 600FE000 len 00001FD8 (7 KiB): RTCRAM
I (156) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (163) spi_flash: detected chip: generic
I (166) spi_flash: flash io: qio
I (170) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (176) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (182) main_task: Started on CPU0
I (192) esp_psram: Reserving pool of 64K of internal memory for DMA/internal allocations
I (192) main_task: Calling app_main()
I (212) Board: UUID=b9e2ec6b-cf06-4bcd-ae1d-b846b9fa7104 SKU=bread-compact-wifi
I (212) button: IoT Button Version: 4.1.3
I (212) CompactWifiBoard: Install SSD1306 driver
I (222) CompactWifiBoard: SSD1306 driver installed
I (222) CompactWifiBoard: Turning display on
I (332) Display: Power management not supported
[... 258 lines cut ...]
I (298433) SystemInfo: free sram: 118911 minimal sram: 107063
I (308433) SystemInfo: free sram: 118431 minimal sram: 107063
I (318433) SystemInfo: free sram: 118431 minimal sram: 107063
I (328433) SystemInfo: free sram: 118859 minimal sram: 107063
I (338433) SystemInfo: free sram: 118431 minimal sram: 107063
I (348433) SystemInfo: free sram: 118947 minimal sram: 107063
I (358433) SystemInfo: free sram: 118395 minimal sram: 107063
I (368433) SystemInfo: free sram: 118947 minimal sram: 107063
I (378433) SystemInfo: free sram: 118435 minimal sram: 107063
I (388433) SystemInfo: free sram: 118911 minimal sram: 107063
I (398433) SystemInfo: free sram: 118431 minimal sram: 107063
I (408433) SystemInfo: free sram: 118431 minimal sram: 107063
I (418433) SystemInfo: free sram: 118915 minimal sram: 107063
I (428433) SystemInfo: free sram: 118435 minimal sram: 107063
I (438433) SystemInfo: free sram: 118947 minimal sram: 107063
I (448433) SystemInfo: free sram: 118431 minimal sram: 107063
I (458433) SystemInfo: free sram: 118947 minimal sram: 107063
I (468433) SystemInfo: free sram: 118431 minimal sram: 107063
I (478433) SystemInfo: free sram: 118431 minimal sram: 107063
I (488433) SystemInfo: free sram: 118971 minimal sram: 107063
I (498433) SystemInfo: free sram: 118415 minimal sram: 107063
I (508433) SystemInfo: free sram: 118971 minimal sram: 107063
I (518433) SystemInfo: free sram: 118451 minimal sram: 107063
I (528433) SystemInfo: free sram: 118931 minimal sram: 107063
`

Xuantam91, 2025-09-21:
After flash firmware, my device is running normal. But it looks like it has a problem with the microphone. I call I Lily it isn't wake, I press boot it alway show listening... and not recognize any things when I say promt 
How do i check and solve this?

Xuantam91, 2025-09-22:
I found an abnormal when I flash the FW in VSCode with Esp Idf extenion. After the flash is done, my device connects to my wifi automatically. I don't need to go to the Ap wifi and config. So I think it is only OTA flash, still remaining old SW.
Can you help to teach me, build merged.bin, and flash totally fw to my device?

78, 2025-09-24:
> [@78](https://github.com/78) can you help to advice on this case?

If you want to further develop the device, it would be best to request the schematic from your seller.

closed 2026-10-21 by xuantam91